PhilipsHue/flutter_reactive_ble #798: Feature to know if still subscribed to characteristic notification.

vicasystem, 2023-09-18:
Hello,
I have some problems on Android (iPhone works correctly), where after subscribing to a characteristic the OS immediately close the subscription. What I see in the log is this:

D/BluetoothGatt( 2631): setCharacteristicNotification() - uuid: 00006001-0000-1000-8000-00805f9b34fb enable: **true**
and immediately after:
D/BluetoothGatt( 2631): setCharacteristicNotification() - uuid: 00006001-0000-1000-8000-00805f9b34fb enable: **false**

The problem is that I didn't find any way to check the subscription status in order to resubscribe to the characteristic.
Is there any way to know if subscribe is still ok?
Or any idea why I have this problem?

Thanks.

remonh87, 2023-09-27:
If subscribing fails normally the stream is automatically cancelled. As long as the stream is active you subscribed to an characteristic.

Taym95, 2024-05-22:
I will close this because it is not library issues

closed 2024-05-22 by Taym95