PhilipsHue/flutter_reactive_ble #799: Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.}

samnbd, 2023-09-23:
**Describe the bug**
Getting an error when trying to pair to BLE device on iOS: 
Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.}

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one devices using `connectTo()` 

**Expected behavior**
Expected to pair to our device.

**Smartphone / tablet**
 - Device:  iPhone
 - OS:  iOS 13 and above
 - Package version: 5.2.0

**Peripheral device**
- Vendor, model: Pavlok 3
- Does it run a custom firmware / software: yes 

**Additional context**
So, the device pairs without issues on my iPhone but in production it fails to pair on most iOS device with the error mentioned above.
PS: This issue is only happening on iOS

chipweinberger, 2023-11-06:
this means your ble device stopped working

samnbd, 2023-11-06:
well no, we have a lot of users having the same problem and the device isn't stopping at all. In fact when they attempt to pair again, it connects correctly. This is is randomly happening to most of our iOS users

chipweinberger, 2023-11-06:
sure that may be true, but from the perspective of iOS, the device stopped working. 

Your device is constantly responding to the platform with its current status. 

If those status updates do not happen quickly enough, iOS will assume your device is malfunctioning.

samnbd, 2023-11-06:
Here are some resources for this issue reported by other users:

- [https://developer.apple.com/forums/thread/122325](https://developer.apple.com/forums/thread/122325)
- [https://lists.apple.com/archives/bluetooth-dev/2013/Apr/msg00154.html](https://lists.apple.com/archives/bluetooth-dev/2013/Apr/msg00154.html)

Gufdoor, 2024-02-23:
Having the same issue here. I have a custom device which uses a NINA b302-00b for BLE connection and an iPhone XR, 11 and 15, all with iOS 17.3.1  (tho the issue happened to previous versions too). Every time I try a new connection it ends in time out, I also tested using nRF Connect app, not successful. Any news?

Taym95, 2024-05-22:
This is not library issue but ble device issues, I will close this

closed 2024-05-22 by Taym95