PhilipsHue/flutter_reactive_ble #809: iOS will not connect to peripheral that removed the bond information.

delivertango, 2023-10-23:
**Describe the bug**
If iOS is bonded with a peripheral and the peripheral removes the bond, iOS will not connect until you remove the peripheral from the bluetooth settings.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the iPhone to the peripheral using ‘connectToDevice()’
2. Peripheral start bonding process with the iPhone
3. Remove bond information in the peripheral
4. Connect again to the peripheral using ‘connectToDevice()’

**Expected behavior**
I would like to be able to remove the peripheral from the bluetooth devices or start the bonding process again.

**Smartphone / tablet**
 - Device: iPhone 8 Plus
 - OS: iOS 16.7.1
 - Package version: 5.2.0

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware / software: yes

**Additional context**
I'm developing a device with bluetooth (peripheral) and an Android/iPhone app to control it using this package.
The user can factory reset the peripheral, and that will lead to remove all the bonding information.
Android handles this well, it seems to check that the bond was removed and forget it from the bluetooth devices so the peripheral can start the bonding process again.
Meanwhile, iOS seems to try to connect but will not start the bonding process again, just disconnects.
The process to remove it from the settings can be hard for some people without knowledge about smartphones, and I would like to know if it’s posible to remove the bond from the iPhone or re-bond without removing it using the package.

BenelliFurtado, 2023-11-06:
Hi, 
according to these posts, it seems that this is intended on iOS and that there is no native way to check if the bond was deleted intentionally:
https://devzone.nordicsemi.com/f/nordic-q-a/31011/ios-and-nrf51822-pairing-after-erasing-bonding-info-problem

(But if someone finds a way to do it, I would also be interested to know!)

enseitankad0, 2024-02-13:
Have you found a solution to this problem? I'm facing the same problem. After the DFU update, this situation can be reproduced on iOS easily. I described my issue [here.](https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/353#issuecomment-1938854219)

It is possible to manually remove bonding by going to settings, and you'll probably resolve this problem.

Taym95, 2024-05-22:
We cannot control IOS Bonding from library, this is limitation!

closed 2024-05-22 by Taym95