PhilipsHue/flutter_reactive_ble #813: CBATTErrorDomain:10, The attribute could not be found

nguyenhuuta, 2023-10-27:
**Describe the bug**
Unhandled Exception: PlatformException(CBATTErrorDomain:10, The attribute could not be found., {NSLocalizedDescription: The attribute could not be found.},

**To Reproduce**
Steps to reproduce the behavior:
1. scanForDevices -> OK
2. connectToDevice -> OK
3. discoverAllServices -> OK
4. characteristic.subscribe -> error

**Expected behavior**
A clear and concise description of what you expected to happen.

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Iphone 13 Pro max
 - OS: 16.3.1
 - Package version: 5.2.0

Android still works fine, I see there is a difference between android and ios regarding UUID 
[image]
[image]

Taym95, 2023-11-01:
Android publishes as Service UUID as long ID, iOS publishes as Service UUID as short ID, try this for example:

`  static const _shortUuid = "ffff";
  static const _extendedUuid = "0000ffff-0000-9999-9999-0099999999";
  static final Uuid uuid = Platform.isAndroid ? Uuid.parse(_extendedUuid) : Uuid.parse(_shortUuid);`

closed 2023-11-01 by Taym95

nguyenhuuta, 2023-11-04:
Thanks you for reply, So what about the problem "characteristics.subscribe -> error"