PhilipsHue/flutter_reactive_ble #819: App crash - exception could not be delivered

pmagnuson, 2023-11-11:
**Describe the bug**
When the Bluetooth device disconnects, the Bluetooth stack has an undeliverable exception and the app crashes.

**To Reproduce**
So far, this has only happened on Android, not on iOS.

Steps to reproduce the behavior:
1. Connect to one device using `connectTo()`
2. Write to a protected characteristic
3. Read a characteristic which reports the bonding status
4. Once bonding is complete, write app specific commands and setup notify streams on 2 characteristics
5. The device does a Bluetooth disconnect and 5-10 seconds later starts advertising again. 
6. 10-20% of the time flutter_reactive_ble will encounter the exception and the library will crash.
7. 80-90% of the time the device starts advertising again and the app successfully scans for the device.  Repeat from step 1.

**Expected behavior**
The Bluetooth library should not crash and cause the app to crash.  

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Google Pixel 5a
 - OS: Android 13 SDK 33
 - Package version: 5.0.2
 - 
**Also on**
 - Device: Samsung Galaxy S22
 - OS: Android 12
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware / software: yes 
- Device runs Zephyr OS and uses the Zephyr/Nordic Bluetooth libraries

**Additional context**
In the following logs, lines with `[log]` are from the application.

**successful connection**
```log
I/flutter (30524): REACTIVE_BLE: Received ScanResult(result: Result<DiscoveredDevice, GenericFailure<ScanFailure>?>.success(DiscoveredDevice(id: 52:A6:D4:89:11:96, name: DharmaDr, serviceData: {}, serviceUuids: [0eaa6289-88f3-4817-985e-f0b869ad5e05], manufacturerData: [], rssi: -64, connectable: Connectable.available)))
[log] FINER: 14:43:53:777: TabProvider: new Device 52:A6:D4:89:11:96 DharmaDr
I/flutter (30524): REACTIVE_BLE: Received ScanResult(result: Result<DiscoveredDevice, GenericFailure<ScanFailure>?>.success(DiscoveredDevice(id: 52:A6:D4:89:11:96, name: DharmaDr, serviceData: {}, serviceUuids: [0eaa6289-88f3-4817-985e-f0b869ad5e05], manufacturerData: [], rssi: -69, connectable: Connectable.available)))
[log] FINER: 14:43:53:780: TabProvider: leftConnectDevice 52:A6:D4:89:11:96
[log] FINER: 14:43:53:782: TabProvider: new Device 52:A6:D4:89:11:96 DharmaDr
I/flutter (30524): REACTIVE_BLE: Connect to device: 52:A6:D4:89:11:96, servicesWithCharacteristicsToDiscover: null, timeout: 0:00:01.500000
I/flutter (30524): REACTIVE_BLE: Received ConnectionStateUpdate(deviceId: 52:A6:D4:89:11:96, connectionState: DeviceConnectionState.connecting, failure: null)
[log] FINER: 14:43:54:344: TabProvider: leftConnectDevice 52:A6:D4:89:11:96 connecting null
D/BluetoothGatt(30524): connect() - device: 52:A6:D4:89:11:96, auto: false
D/BluetoothGatt(30524): registerApp()
D/BluetoothGatt(30524): registerApp() - UUID=8893b941-7ead-4497-bf04-2f86ec9907bd
D/BluetoothGatt(30524): onClientRegistered() - status=0 clientIf=7
[log] FINER: 14:43:54:409: TabProvider: new Device 52:A6:D4:89:11:96 DharmaDr
I/flutter (30524): REACTIVE_BLE: Received ScanResult(result: Result<DiscoveredDevice, GenericFailure<ScanFailure>?>.success(DiscoveredDevice(id: 52:A6:D4:89:11:96, name: DharmaDr, serviceData: {}, serviceUuids: [0eaa6289-88f3-4817-985e-f0b869ad5e05], manufacturerData: [], rssi: -65, connectable: Connectable.available)))
D/BluetoothGatt(30524): onClientConnectionState() - status=0 clientIf=7 device=52:A6:D4:89:11:96
I/flutter (30524): REACTIVE_BLE: Received ConnectionStateUpdate(deviceId: 52:A6:D4:89:11:96, connectionState: DeviceConnectionState.connected, failure: null)
[log] FINER: 14:43:54:855: TabProvider: leftConnectDevice 52:A6:D4:89:11:96 connected null
[log] FINER: 14:43:54:856: AppBle: writeSecureMode
[log] FINEST: 14:43:54:858: AppBle: writeCommandWithResponse 52:A6:D4:89:11:96 0eaa6289 0eaa6294 [254]
I/flutter (30524): REACTIVE_BLE: Get discovered services for device: 52:A6:D4:89:11:96
I/flutter (30524): REACTIVE_BLE: Get discovered services for device: 52:A6:D4:89:11:96
D/BluetoothGatt(30524): discoverServices() - device: 52:A6:D4:89:11:96
D/BluetoothGatt(30524): onSearchComplete() = Device=52:A6:D4:89:11:96 Status=0
I/flutter (30524): REACTIVE_BLE: Write with response to CharacteristicInstance(characteristicId: 0eaa6294-88f3-4817-985e-f0b869ad5e05(34), serviceId: 0eaa6289-88f3-4817-985e-f0b869ad5e05(18), deviceId: 52:A6:D4:89:11:96), value: [254]
I/flutter (30524): REACTIVE_BLE: Read characteristic: CharacteristicInstance(characteristicId: 0eaa6294-88f3-4817-985e-f0b869ad5e05(34), serviceId: 0eaa6289-88f3-4817-985e-f0b869ad5e05(18), deviceId: 52:A6:D4:89:11:96)
D/BluetoothGatt(30524): onPhyUpdate() - status=0 address=52:A6:D4:89:11:96 txPhy=2 rxPhy=2
[log] FINER: 14:43:57:800: TabProvider: app inactive lockScreen: false inLocalMode true
[log] FINER: 14:43:59:008: TabProvider: app inactive lockScreen: false inLocalMode true
[log] FINER: 14:43:59:009: TabProvider: app resumed lockRestart false
D/BluetoothGatt(30524): onConnectionUpdated() - Device=52:A6:D4:89:11:96 interval=24 latency=0 timeout=42 status=0
[... 250 lines cut ...]
E/AndroidRuntime(30524): 	at io.flutter.plugin.common.EventChannel$IncomingStreamRequestHandler.onMessage(EventChannel.java:199)
E/AndroidRuntime(30524): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
E/AndroidRuntime(30524): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:322)
E/AndroidRuntime(30524): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
E/AndroidRuntime(30524): 	at android.os.Handler.handleCallback(Handler.java:942)
E/AndroidRuntime(30524): 	at android.os.Handler.dispatchMessage(Handler.java:99)
E/AndroidRuntime(30524): 	at android.os.Looper.loopOnce(Looper.java:201)
E/AndroidRuntime(30524): 	at android.os.Looper.loop(Looper.java:288)
E/AndroidRuntime(30524): 	at android.app.ActivityThread.main(ActivityThread.java:7918)
E/AndroidRuntime(30524): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime(30524): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
E/AndroidRuntime(30524): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
E/AndroidRuntime(30524): Caused by: com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)
E/AndroidRuntime(30524): 	at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
E/AndroidRuntime(30524): 	at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:299)
E/AndroidRuntime(30524): 	at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:864)
E/AndroidRuntime(30524): 	at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback(Unknown Source:0)
E/AndroidRuntime(30524): 	at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:294)
E/AndroidRuntime(30524): 	at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:128)
E/AndroidRuntime(30524): 	at android.os.Binder.execTransactInternal(Binder.java:1285)
E/AndroidRuntime(30524): 	at android.os.Binder.execTransact(Binder.java:1244)
I/Process (30524): Sending signal. PID: 30524 SIG: 9
Lost connection to device.
Exited
```

Baxi19, 2024-05-30:
Hi @pmagnuson 

To resolve the issue, you primarily need to add the necessary dependency in your `android/app/build.gradle` file, like so:
```shell
android {
    // Other configurations...
}

flutter {
    source '../..'
}

dependencies {
    // TODO: Add this dependency
    implementation "com.polidea.rxandroidble2:rxandroidble:1.11.1"
}
``` 

Next, you'll need to update your  `Android´s main file`. In my case, I'm using `Kotlin` (for Java the solution is in the [README](https://github.com/PhilipsHue/flutter_reactive_ble?tab=readme-ov-file#faq)) , so your file might look something like this:
 ```kotlin
//TODO: Update it with your app information
package com.your_company.your_app

import io.flutter.embedding.android.FlutterActivity
import android.os.Bundle
import com.polidea.rxandroidble2.exceptions.BleException
import io.reactivex.exceptions.UndeliverableException
import io.reactivex.plugins.RxJavaPlugins

class MainActivity: FlutterActivity() {
    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)

        RxJavaPlugins.setErrorHandler { throwable ->
            if (throwable is UndeliverableException && throwable.cause is BleException) {
                // ignore BleExceptions since we do not have subscriber
                return@setErrorHandler
            } else {
                throw throwable
            }
        }
    }
}
``` 

Finally, after making these changes, clean your project to get the dependencies and then run it again. 
 ```shell
flutter clean && flutter pub get && flutter run
 ```

The issue should now be resolved :octocat: 🔨

fourman-alex, 2024-12-10:
@chipweinberger @Baxi19 Are you guys saying that this is not a bug of `flutter_reactive_ble` and instead is an additional setup step of the package on Android? 
In other words, should the installation steps also instruct of adding a `RxJavaPlugins` error handler? 
Or, is there something the can be done in the package's code that will handle those exceptions?