PhilipsHue/flutter_reactive_ble #835: 'readRssi' Method Not Defined for 'ReactiveBlePlatform'

DeToxer, 2024-01-26:
**Describe the bug**
After upgrading the `flutter_reactive_ble` library from version `5.2.0` to `5.3.0,` the app fails to compile on both Android and iOS platforms. The compiler error indicates that the method `readRssi` isn't defined for the class `ReactiveBlePlatform`. This issue has been tested and confirmed on two devices: an iPhone simulator (iOS version 17.0, iPhone 15 Pro Max) and an Android emulator (Pixel 7 Pro API 33). The problem persists despite attempts to clean the build environment using `flutter clean `and `flutter pub cache clean --force`. Notably, this issue does not arise in a newly created Flutter project.

**Smartphone / tablet**
Device: iPhone 15 Pro Max/Pixel 7 Pro
OS: iOS 17.0/Android 12 (SDK 33)
Package version: `flutter_reactive_ble 5.3.0`

**Additional context**
The issue appears to be specifically associated with version 5.3.0 of the `flutter_reactive_ble library`. Reverting to version 5.2.0 or initiating a new Flutter project does not result in this compile-time error, suggesting that the problem is isolated to this particular version update.

Android stack trace:
```
- 'ReactiveBlePlatform' is from 'package:reactive_ble_platform_interface/src/reactive_ble_platform_interface.dart' ('../../../../../.pub-cache/hosted/pub.dev/reactive_ble_platform_interface-5.2.0/lib/src/reactive_ble_platform_interface.dart').
Try correcting the name to the name of an existing method, or defining a method named 'readRssi'.
      _blePlatform.readRssi(deviceId);
                   ^^^^^^^^
Target kernel_snapshot failed: Exception

FAILURE: Build failed with an exception.
```

iOS stack tace:
```
Error (Xcode): ../../../../../.pub-cache/hosted/pub.dev/flutter_reactive_ble-5.3.0/lib/src/reactive_ble.dart:405:20: Error: The method 'readRssi' isn't defined for the class 'ReactiveBlePlatform'.

Could not build the application for the simulator.
Error launching application on iPhone 15 Pro Max.
```

farr64, 2024-01-26:
I ran into the same problem. My solution was to force this in `pubspec.yaml` without the "^"

`flutter_reactive_ble: 5.2.0`

Up tp now, I had it as "^5.2.0" with the "^" (which allowed never versions, such as the ill-fated 5.3.0). 

Now, 5.3.0 is not allowed and things continue to work perfectly with 5.2.0.

Then, I deleted this directory:

`~/.pub-cache/hosted/pub.dev/flutter_reactive_ble-5.3.0/`

farr64, 2024-01-26:
I'm sure the good folks in charge of this wonderful repository will come up with `5.3.1` ASAP. 

Thanks!

ffamar, 2024-01-26:
I have the same issue.

farr64, 2024-01-26:
I'm afraid everyone who goes from `5.2.0` to` 5.3.0` will have the same issue. See my solution above.

DeToxer, 2024-01-26:
I see that the `pubspec.yaml` inside [flutter_reactive_ble](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/packages/flutter_reactive_ble/pubspec.yaml) and [reactive_ble_mobile](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/packages/reactive_ble_mobile/pubspec.yaml) still targets
`reactive_ble_platform_interface: ^5.2.0 `and `reactive_ble_mobile: ^5.2.0` instead of 5.3.0.
That might be the issue.

Taym95, 2024-02-05:
my bad will fix it and make realese soon

closed 2024-02-05 by Taym95