PhilipsHue/flutter_reactive_ble #836: protobuf ^3.1.0 support

nissaba, 2024-01-26:
My project was started suing this lib, and also needs to use mcumgr_flutter.

But I needed to update mcumgr_flutter and other libs because Apple as set the minimum target at iOS 12. 
This cause the project to break because reactive_ble_mobile depends on protobuf ^2.0.0 and mcumgr_flutter >=0.2.0+1 and so mcumgr_flutter >=0.2.0+1 is incompatible with flutter_reactive_ble >=5.0.3, and because MyProject depends on both flutter_reactive_ble ^5.0.3 and mcumgr_flutter ^0.3.0+1, version solving failed

nissaba, 2024-02-14:
ping!

keithle888, 2024-04-27:
This would be useful for me too as protobuf [well known types](https://pub.dev/documentation/protobuf_wellknown/latest/) requires protobuf 3.x. I haven't yet come across another way of solving using custom protobuf messaging while importing well-known types in the .proto schema.

rohansingh, 2024-06-07:
We've been running successfully for quite a while with this workaround:

```yaml
dependency_overrides:
  protobuf: 3.1.0
```