PhilipsHue/flutter_reactive_ble #838: scanForDevices should not use 'assert(!central.isScanning)'

nissaba, 2024-01-30:
In the function  

func scanForDevices(name: String, args: ScanForDevicesRequest, completion: @escaping PlatformMethodCompletionHandler) {
        guard let central = central
        else {
            completion(.failure(PluginError.notInitialized.asFlutterError))
            return
        }

        assert(!central.isScanning)

        scan = StreamingTask(parameters: .init(services: args.serviceUuids.map({ uuid in CBUUID(data: uuid.data) })))

        completion(.success(nil))
    }

you should not use the assert(!central.isScanning) as this will crash an app that is trying connect to a device and wants to move to scaning to all device. 

This is the case for my app. When starting the app I will call connectToAdvertisingDevice to find the last connected device or to reconnect. if the user at some point wants to scan for a new/other device will need to cut the process of this scanForDevices, as there si no way to interupt it asking to discover new devices will lead to a crash.

What you the code do, it should stop/kill the current decovery process and initiate a new request/call.

nissaba, 2024-02-14:
ping!

Taym95, 2024-03-01:
Hi thanks for report, PR are welcome

Albert-Jan, 2024-07-01:
Facing the same issue

i13Proietti, 2024-08-01:
I'm in trouble too with this problem. Waiting for the solution

YueLiXing, 2024-08-16:
I have found a way to bypass this issue

```dart
class ShareReactiveBle {
  static Uuid gattUUID = Uuid.parse('****');
  static Uuid meshUUID = Uuid.parse('****');
  static final ShareReactiveBle _instance = ShareReactiveBle._internal();

  factory ShareReactiveBle() => _instance;

  late final FlutterReactiveBle _ble;
  Stream<DiscoveredDevice>? _stream;

  FlutterReactiveBle get ble => _ble;

  ShareReactiveBle._internal() {
    _ble = FlutterReactiveBle();
  }

  Stream<DiscoveredDevice> scan() {
    final filterService = Platform.isIOS ? [gattUUID, meshUUID] : <Uuid>[];
    _stream ??= _ble.scanForDevices(withServices: filterService, scanMode: ScanMode.lowLatency).asBroadcastStream();
    return _stream as Stream<DiscoveredDevice>;
  }
}
```