PhilipsHue/flutter_reactive_ble #839: A problem occurred evaluating project ':reactive_ble_mobile'.

KonstantinidouCh, 2024-01-31:
- While my app is building fine on vs code, when I am trying to build an apk for my flutter app, I get the following error : 

FAILURE: Build failed with an exception.

* Where:
Build file '/root/.pub-cache/hosted/pub.dev/reactive_ble_mobile-5.3.0/android/build.gradle' line: 40

* What went wrong:
A problem occurred evaluating project ':reactive_ble_mobile'.
> No signature of method: build_45yl2so7zx8yhlv0dwd4gmvrz.android() is applicable for argument types: (build_45yl2so7zx8yhlv0dwd4gmvrz$_run_closure2) values: [build_45yl2so7zx8yhlv0dwd4gmvrz$_run_closure2@20f8d4a7]

- I tried to downgrade flutter_reactive_ble to 5.2.0 but I still get the same error. 

- My project is using flutter 3.13.2 and kotlin 1.8.21

andrepura, 2024-01-31:
i have the same issue here
`
FAILURE: Build failed with an exception.

* Where:
Build file 'C:\Users\AndrePura\AppData\Local\Pub\Cache\hosted\pub.dev\reactive_ble_mobile-5.3.0\android\build.gradle' line: 40

* What went wrong:
A problem occurred evaluating project ':reactive_ble_mobile'.
> No signature of method: build_ahn2lhvy9c9apm415rd8cuhg0.android() is applicable for argument types: (build_ahn2lhvy9c9apm415rd8cuhg0$_run_closure2) values: [build_ahn2lhvy9c9apm415rd8cuhg0$_run_closure2@616bb527]

`

`flutter_reactive_ble: ^5.3.0`
`compileSdkVersion 33`
`Flutter 3.16.9`

andrepura, 2024-02-01:
i fixed the problem by updating to newer versions in build.gradle (see: kotlin_version && build:gradle)

```
buildscript {
    ext.kotlin_version = '1.8.22'
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:7.0.2'
        classpath "org.jetbrains.kotlin:kotlin-gradle-plugin:$kotlin_version"
    }
}
```
and updateing gradle-wrapper.properties in

`distributionUrl=https\://services.gradle.org/distributions/gradle-7.0.2-all.zip`

could be possibile that newer gradle versions will work but i haven't tested it yet.

i also updated my sdk command line tools (they were out of sync with the android studio version)

KonstantinidouCh, 2024-02-01:
> i fixed the problem by updating to newer versions in build.gradle (see: kotlin_version && build:gradle)
>
> ```
> buildscript {
>     ext.kotlin_version = '1.8.22'
>     repositories { [...]

That seems to solve my problem! Thank you!

closed 2024-02-01 by KonstantinidouCh