PhilipsHue/flutter_reactive_ble #841: compileSdkVersion is not specified

jyk2joe, 2024-02-02:
I encountered the following error when compiling

* What went wrong:
A problem occurred configuring project ':reactive_ble_mobile'.
> Failed to notify project evaluation listener.
   > Cannot invoke method substring() on null object
   > compileSdkVersion is not specified. Please add it to build.gradle

vbuberen, 2024-02-05:
You could at least look at the source code to see that compileSDK is specified in that package https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/packages/reactive_ble_mobile/android/build.gradle#L41

So it is something on your side.

Taym95, 2024-02-05:
As @vbuberen said: `compileSdkVersion 33` already specified

closed 2024-02-05 by Taym95

haxzzallian, 2026-01-17:
if app_links package is part of error because several packages requires it underground, you can add this to the pub.yaml file " dependency_overrides:
  app_links: ^5.0.0  ". note you can change the version to ascertain which version works with your code