PhilipsHue/flutter_reactive_ble #854: BLE IOS Running on Background Mode is unstable

sangtran1711, 2024-03-20:
**Describe the bug**
My app try to read data from a thermometer device, the device advertises every 3, 6, 9 or 12 minutes based on device calculation and my app sometime didn't receive data for 1-2 hours or more.

**To Reproduce**
Steps to reproduce the behavior:
App run 
1. Connect to device using `connectToDevice()`
2. Read from characteristic
3. Disconnect
4. Reconnect to device again.

**Expected behavior**
My expectation is my app receive data from the device every 3, 6, 9 or 12 minutes at least.

**Smartphone / tablet**
 - Device: iPhone 11 Pro Max
 - OS: iOS 16
 - Package version: 5.0.3

**Peripheral device**
- Vendor, model: Thermometer device

**Additional context**
My app flow is Connect to the device -> read data -> disconnect -> wait 2 seconds -> reconnect to the device again. This will be forever until the user unregister the device from the app. The app will run 1 or several days smoothly as I expected and then it started not receiving data for 1-2 hours and then received data again. Don't know if it is an issue or a limitation of IOS.

Scottie-Fischer, 2024-05-25:
This could be because IOS can suspend applications. Without a restoration key and option for restoration:
`CBCentralManager( delegate: centralManagerDelegate, queue: nil, options:  [ CBCentralManagerOptionRestoreIdentifierKey: centralManagerRestoreID, CBCentralManagerOptionShowPowerAlertKey: true ] )`
the app could remain suspended even if the bluetooth connection receives an update. Furthermore, if the app is suspended the connection could be severed while in the background. Honestly this restoration should be an added feature.

drolpi, 2024-07-03:
@Taym95 are there any updates on this?