PhilipsHue/flutter_reactive_ble #856: Add privacy manifest for Apple

tony-ditchlabs, 2024-03-22:
See https://github.com/flutter/flutter/issues/131940

Apple announcement here: https://developer.apple.com/support/third-party-SDK-requirements/

May 1st deadline announcement here: https://developer.apple.com/news/?id=3d8a9yyh

Taym95, 2024-03-25:
Thanks @tony-ditchlabs

truedat101, 2025-02-19:
This appears to be a real problem right now, related specifically to protobuf dependency (now, as in this started being a problem on the 12th of Feb).

From Apple test flight in my app submission:
```
ITMS-91061: Missing privacy manifest - Your app includes “Frameworks/Protobuf.framework/Protobuf”, which includes Protobuf, an SDK that was identified in the documentation as a commonly used third-party SDK. If a new app includes a commonly used third-party SDK, or an app update adds a new commonly used third-party SDK, the SDK must include a privacy manifest file. Please contact the provider of the SDK that includes this file to get an updated SDK version with a privacy manifest. For more details about this policy, including a list of SDKs that are required to include signatures and manifests, visit: https://developer.apple.com/support/third-party-SDK-requirements. 
```
This dependency is coming from flutter_reactive_ble

truedat101, 2025-02-19:
More details, it comes from

```
reactive_ble_mobile 5.4.0
│   │   ├── **_protobuf 2.1.0_**
│   │   │   ├── fixnum 1.1.0
│   │   │   ├── collection...
│   │   │   └── meta...
│   │   ├── flutter...
│   │   └── reactive_ble_platform_interface...
```