PhilipsHue/flutter_reactive_ble #862: Supported Devices

phildharmadr, 2024-05-07:
**Describe the bug**
We have a functioning app that works across iPhone7, iPhone7 plus, iPhone 12, iPhone14, Pixel 5, Pixel 8, Samsung Galaxy S22 and operates with our BLE device.

Recently we ran into a Samsung Galaxy S20 with SDK33, Android 13 and we are unable to connect to our BLE device.  

Is there a list of supported devices and OS versions?

pmagnuson, 2024-05-09:
the log capture below shows a little bit of information about this failure.
`[log]` entries are from the application logging (with timestamp)
`I/Bluetooth` entries are from the verbose logging on the ble library.

From the application perspective
- connected status occurs at  08:46:58.983
- a disconnect is received at  08:47:04.624
This sequence occurs every time and is consistently a 5 second interval.  
In this log the application does a write to characteristic 0eaa6294-88f3-4817-985e-f0b869ad5e05.  However the disconnect occurs even when the write is not done.  

For this application and device, this error only occurs on the Galaxy S20 and not other iOS or Android devices that have been tested.  

```log
[log] 08:46:27.091: TabProvider: startDiscovery
I/flutter (22327): REACTIVE_BLE: Scan for devices with services:[0eaa6289-88f3-4817-985e-f0b869ad5e05, 8d53dc1d-1db7-4cd3-868b-8a527460aa84], scanMode: ScanMode.balanced, requireLocationServicesEnabled: true
[log] 08:46:27.116: TabSessionCountProvider: user brieS366tsh0CCMkYCu1b0Crpbf1 usageCount 188
I/BLASTBufferQueue(22327): update, w= 1080 h= 2400 mName = ViewRootImpl@c62eca9[MainActivity] mNativeObject= 0xb4000075eb9dce70 sc.mNativeObject= 0xb40000757ba34f70 format= -3 caller= android.view.ViewRootImpl.updateBlastSurfaceIfNeeded:2898 android.view.ViewRootImpl.relayoutWindow:9847 android.view.ViewRootImpl.performTraversals:3884 android.view.ViewRootImpl.doTraversal:3116 android.view.ViewRootImpl$TraversalRunnable.run:10885 android.view.Choreographer$CallbackRecord.run:1301
[log] 08:46:27.117: DatabaseService: checkUserDebugLevel not found
[log] 08:46:27.117: DatabaseService: getUserInfo logLevel FINER
[log] 08:46:27.117: cloudLogging: updateUserLogLevel FINER
[log] 08:46:27.117: DatabaseService: getUserInfo brieS366tsh0CCMkYCu1b0Crpbf1 {image: https://lh3.googleusercontent.com/a/ACg8ocL1OkNgDEsG3gtRXbvq6R12OCjPhrSi9gbWZIG6bLaJAjq9=s96-c, SAVETHREAD: thread_qEQP54daRj42QHXnZRIiiFVd, session_count: 188, name: phil magnuson, last_active: Timestamp(seconds=1715265986, nanoseconds=252000000), email: phillmag@gmail.com}
I/BluetoothAdapter(22327): STATE_ON
I/BluetoothAdapter(22327): STATE_ON
I/BluetoothAdapter(22327): STATE_ON
I/BluetoothAdapter(22327): STATE_ON
D/BluetoothLeScanner(22327): Start Scan with callback
D/BluetoothLeScanner(22327): onScannerRegistered() - status=0 scannerId=4 mScannerId=0
I/flutter (22327): REACTIVE_BLE: Received ScanResult(result: Result<DiscoveredDevice, GenericFailure<ScanFailure>?>.success(DiscoveredDevice(id: 76:6C:68:B5:8A:54, name: DharmaDr, serviceData: {}, serviceUuids: [0eaa6289-88f3-4817-985e-f0b869ad5e05], manufacturerData: [], rssi: -73, connectable: Connectable.available)))
[log] 08:46:58.177: TabProvider: discovery 76:6C:68:B5:8A:54 DharmaDr inProgress false
[log] 08:46:58.179: TabProvider: connect start 76:6C:68:B5:8A:54
I/flutter (22327): REACTIVE_BLE: Received ScanResult(result: Result<DiscoveredDevice, GenericFailure<ScanFailure>?>.success(DiscoveredDevice(id: 76:6C:68:B5:8A:54, name: DharmaDr, serviceData: {}, serviceUuids: [0eaa6289-88f3-4817-985e-f0b869ad5e05], manufacturerData: [], rssi: -73, connectable: Connectable.available)))
[log] 08:46:58.247: TabProvider: discovery 76:6C:68:B5:8A:54 DharmaDr inProgress true
I/flutter (22327): REACTIVE_BLE: Connect to device: 76:6C:68:B5:8A:54, servicesWithCharacteristicsToDiscover: null, timeout: 0:00:01.500000
I/BluetoothAdapter(22327): STATE_ON
D/BluetoothGatt(22327): connect() - device: 766C68_4, auto: false
I/flutter (22327): REACTIVE_BLE: Received ConnectionStateUpdate(deviceId: 76:6C:68:B5:8A:54, connectionState: DeviceConnectionState.connecting, failure: null)
D/BluetoothGatt(22327): registerApp()
D/BluetoothGatt(22327): registerApp() - UUID=54f40e04-01cf-47e5-80de-da66c94a6395
D/BluetoothGatt(22327): onClientRegistered() - status=0 clientIf=6
[log] 08:46:58.725: TabProvider: connect listen 76:6C:68:B5:8A:54 connecting null
D/BluetoothGatt(22327): onClientConnectionState() - status=0 clientIf=6 device=766C68_4
I/flutter (22327): REACTIVE_BLE: Received ConnectionStateUpdate(deviceId: 76:6C:68:B5:8A:54, connectionState: DeviceConnectionState.connected, failure: null)
[log] 08:46:58.983: TabProvider: connect listen 76:6C:68:B5:8A:54 connected null
[log] 08:46:58.985: AppBle: writeSecureMode
I/flutter (22327): REACTIVE_BLE: Get discovered services for device: 76:6C:68:B5:8A:54
I/flutter (22327): REACTIVE_BLE: Get discovered services for device: 76:6C:68:B5:8A:54
D/BluetoothGatt(22327): discoverServices() - device: 766C68_4
D/BluetoothGatt(22327): onConnectionUpdated() - Device=766C68_4 interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(22327): onPhyUpdate() - status=0 address=766C68_4 txPhy=2 rxPhy=2
D/BluetoothGatt(22327): onClientConnectionState() - status=0 clientIf=6 device=766C68_4
D/BluetoothGatt(22327): close()
D/BluetoothGatt(22327): unregisterApp() - mClientIf=6
[log] 08:47:04.624: TabProvider: connect listen 76:6C:68:B5:8A:54 disconnected ConnectionError.failedToConnect
I/flutter (22327): REACTIVE_BLE: Received ConnectionStateUpdate(deviceId: 76:6C:68:B5:8A:54, connectionState: DeviceConnectionState.disconnected, failure: GenericFailure<ConnectionError>(code: ConnectionError.failedToConnect, message: "Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 0 (GATT_SUCCESS)"))
[... cut ...]
[log] 08:47:04.701: BatteryStatus: connected: false levels: [100, 100]
[log] 08:47:04.706: BatteryStatus: connected: false levels: [100, 100]
I/flutter (22327): REACTIVE_BLE: Received ConnectionStateUpdate(deviceId: 76:6C:68:B5:8A:54, connectionState: DeviceConnectionState.disconnected, failure: null)
I/TRuntime.CctTransportBackend(22327): Making request to: https://firebaselogging-pa.googleapis.com/v1/firelog/legacy/batchlog
D/TrafficStats(22327): tagSocket(6) with statsTag=0xffffffff, statsUid=-1
I/TRuntime.CctTransportBackend(22327): Status Code: 200
```

ThomasWDev, 2024-06-04:
Is this issue resolved?

pmagnuson, 2024-06-04:
This can be closed.  We found that the Galaxy S20 did not respond well to request from the device for the 2M PHY.

It would still be very useful to know what devices are used for testing of the library.