PhilipsHue/flutter_reactive_ble #879: Bluetooth not work in Android 13

Pedro-Serer, 2024-07-10:
I'm called Pedro and i'm using the package flutter_reactive_ble but i'm have some problems.

The package not work in versions above the Android 12.

Firstly i had problems of permissions with the package. Then i changed my manifest file implementing the line:

[image]

The permission problem has stopped but now i have other problem. My Bluetooth start, scan but not connect... What i can make for this to work?

My function:

[image]

The application was supposed to request these permissions as soon as the APP starts

keval-devani, 2024-07-15:
@Pedro-Serer please add BLUETOOTH_CONNECT permission in android manifest for Android API 31+.  

`<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />`

See [here](https://stackoverflow.com/a/75017408/15161717)

Pedro-Serer, 2024-07-15:
@keval-devani these are my permissions that were already in the manifest file:
[image]

Taym95, 2024-08-13:
This is not library issue, works on Android 13 .

closed 2024-08-13 by Taym95