PhilipsHue/flutter_reactive_ble #880: Error scanning devices in the background

franciscogodoy97, 2024-07-17:
When the app goes to the background, I am using the background_fetch dependency, and when it tries to scan I get the following error. This only happens on Android, it works correctly on iOS.

Error: Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Scan failed because application registration failed (code 6)")

config androidManifest.xml

<uses-permission android:name="android.permission.BLUETOOTH" />
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true" />
<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />

thx.

benoit-rolandeau-act, 2024-09-12:
I think your problem may be linked to this: https://developer.android.com/develop/background-work/services/foreground-services

Is your problem happened on Android 14 only?

benoit-rolandeau-act, 2024-09-13:
There is also a linked subject on the external Android Library used to connect to BLE: https://github.com/dariuszseweryn/RxAndroidBle/wiki/Tutorial:-Background-operation

benoit-rolandeau-act, 2024-09-13:
I had the same problem even if I succeeded to manage the foreground services. I have search a long time on internet and I have found this: https://stackoverflow.com/questions/48077690/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0

I have added an advertised UUID in the `scanForDevices` method:

```
flutterBle
          .scanForDevices(
            withServices: [
              Uuid.parse("669a0c20-0008-a7ba-e311-0685c0f7978a")
            ], //BleServiceHelper.getAvailableServices(),
            scanMode: _currentScanMode!,
          )
          .timeout(
            scanTimeoutWithoutEvent,
            onTimeout: _onScanTimeout,
          )
          .listen(
            _addScanDeviceCallback,
            onError: _onScanError,
          );
```

And now it's working on Android 14 and 13.

GoodOldBatman, 2024-10-01:
From my experience with background scanning (without background_fetch) I can confirm that the UUID filter is required to make it work across both iOS & Android (all versions) so I would assume the same is true when using background_fetch.

This is more a question out of curiosity: @franciscogodoy97 were you able to scan on iOS using background_fetch **without** filtering the UUID and still receive scan results?

franciscogodoy97, 2024-11-21:
I tried, and apparently it doesn't work anymore. My issue is that I'm integrating a tag system where, from the background, I update my device's position. The problem is that other customers may have their tag, but from what I see, I need the UUID to scan it and update its position. This is done in case the customer has lost it and wants to know where it is.

gabrielgarciagava, 2025-01-03:
@franciscogodoy97 not clear to me, now you are facing problem with both Android and iOS? If so, maybe you have some misconfiguration with the background_fetch dependency?