PhilipsHue/flutter_reactive_ble #882: [iOS] PlatformException on subscribeToCharacteristic | Device connects successfully.

devtyagi-c, 2024-09-05:
**Describe the bug**

**(Works fine on Android)**

I perform these operations on iOS:
- Connect to Device, move to a new screen
- Click on start monitoring button
- On the click of the button the following operations are performed
    1. Enable notifications on characteristic 86f66001-f706-58a0-95b2-1fb9261e4dc7
    2. Enable notifications on characteristic 86f65002-f706-58a0-95b2-1fb9261e4dc7
    3. Write 0x0213C87462 on characteristic 86f65001-f706-58a0-95b2-1fb9261e4dc7
    4. Write 0x66 on characteristic 86f65001-f706-58a0-95b2-1fb9261e4dc7
  

And I get this error: 
```
PlatformException (PlatformException(reactive_ble_mobile.Central.(unknown context at $10590a818).Failure:1, The operation couldn’t be completed. (reactive_ble_mobile.Central.(unknown context at $10590a818).Failure error 1.), {}, null))
```

[image]

[image]

Is there any fix for this?

**To Reproduce**
Steps to reproduce the behavior:
Mentioned above ^

**Expected behavior**
I should not get this exception, it works perfectly fine on Android.

- [x] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: [e.g. iPhone 13]
 - OS: iOS 17.5.1
 - Package version: ^5.3.1

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware / software: yes

**Additional context**

Here is my code:

Here is my code for the start monitoring button:
```dart
Future<void> startMonitoring() async {
    if (state.selectedDeviceId == null) {
      emit(state.copyWith(error: "No device selected"));
      return;
    }

    if (state.connectionState != DeviceConnectionState.connected) {
      emit(state.copyWith(error: "Device not connected"));
      return;
    }

    // Emit initial state
    emit(state.copyWith(isSettingUpMonitoring: true, error: null));

    try {
      await Future.delayed(const Duration(seconds: 2));

      // Enable notifications on 6001
      _monitorSubscription6001 = _monitorCharacteristic
          .execute(
        state.selectedDeviceId!,
        "86f66000-f706-58a0-95b2-1fb9261e4dc7",
        "86f66001-f706-58a0-95b2-1fb9261e4dc7",
      )
          .listen(
        (data) {
          // Emit state when a new value is received from 6001
          log("Received data from 6001: $data");
          emit(state.copyWith(characteristicValue: data.toString(), isMonitoring: true, isSettingUpMonitoring: false));
        },
        onError: (error) {
          emit(state.copyWith(error: error.toString(), isMonitoring: false, isSettingUpMonitoring: false));
        },
      );

      await Future.delayed(const Duration(milliseconds: 200));

      // Enable notifications on 5002
      _monitorSubscription5002 = _monitorCharacteristic
          .execute(
        state.selectedDeviceId!,
        "86f65000-f706-58a0-95b2-1fb9261e4dc7",
        "86f65002-f706-58a0-95b2-1fb9261e4dc7",
      )
          .listen(
        (data) {
          // We do not need to emit the state here when a new value is received from 5002
          log("Received data from 5002: $data");
        },
        onError: (error) {
          emit(state.copyWith(error: error.toString(), isMonitoring: false, isSettingUpMonitoring: false));
        },
      );

      await Future.delayed(const Duration(milliseconds: 200));

      await _writeCharacteristic.execute(
        state.selectedDeviceId!,
        "86f65000-f706-58a0-95b2-1fb9261e4dc7",
        "86f65001-f706-58a0-95b2-1fb9261e4dc7",
        [0x02, 0x13, 0xC8, 0x74, 0x62],
      );

      log("Successfully wrote data to 86f65001-f706-58a0-95b2-1fb9261e4dc7");

      await _writeCharacteristic.execute(
        state.selectedDeviceId!,
        "86f65000-f706-58a0-95b2-1fb9261e4dc7",
        "86f65001-f706-58a0-95b2-1fb9261e4dc7",
        [0x66],
      );

      log("Successfully wrote data to 86f65001-f706-58a0-95b2-1fb9261e4dc7");
    } catch (e) {
      // Emit state when an error occurs
      log("Error: $e");
      emit(state.copyWith(
        error: e.toString(),
        isMonitoring: false,
        isSettingUpMonitoring: false,
      ));
    }
  }
```

its-me-piyush, 2024-12-09:
any updates on this?

devtyagi-c, 2024-12-09:
@its-me-piyush nope, I moved to Native Apps on iOS and Android.