PhilipsHue/flutter_reactive_ble #898: [Android] More detailed diagnostic for a failed connection

farr64, 2025-03-14:
**Is your feature request related to a problem? (please describe)**

After having worked just fine for months, a particular Android device now refuses to connect. and my log reports this useless jump from `connecting` to  `disconnected` -- without any reason(s) for the failure to connect:

```
I/flutter (10349):  Thu 250313 20:37:08.730 DeviceConnectionState.connecting
I/flutter (10349):  Thu 250313 20:37:08.730 Connecting: abb9cd
I/flutter (10349):  Thu 250313 20:37:08.734 DeviceConnectionState.disconnected
```

I suspect it's a Bluetooth stack problem but, after having tried to reset everything related to Bluetooth, this device continues to refuse to connect.

Other Android devices work perfectly with the same software, which uses `flutter_reactive_ble`
            
With a successful connection, my logs report an additional line produced by `D/BluetoothGatt(25378)` with this useful status report (even though the report for "success" is not really necessary given that a report for "failure" is vitally necessary).
**onClientConnectionState() - status=0** 

```
I/flutter (25378):  Thu 250313 21:35:18.411 DeviceConnectionState.connecting
I/flutter (25378):  Thu 250313 21:35:18.412 Connecting: abb9cd
D/BluetoothGatt(25378): onClientConnectionState() - status=0 clientIf=8 device=D4:DE:07:AB:B9:CD
I/flutter (25378):  Thu 250313 21:35:25.938 DeviceConnectionState.connected
```

- - -
Question: Is there a setting somewhere to get some report of WHY a connection fails?

Something like this would be useful:

**onClientConnectionState() - status=XYZZY** 

with some kind of code that would provide guidance towards some action(s) to fix the fundamental reason(s) for the failure to connect..

            

**Describe the solution you'd like**

Some setting somewhere to get more detailed diagnostic for the specific reason(s) for a failed connection.

**Additional context**

This is the specific Android device that started to fail after it worked fine for months:            
            
brand: Samsung
model: SM-A146U
Android version: 14
API: 34

Many other Android devices work just fine. This is just one example:

brand: BLU
model: G50
Android version: 10
API: 29

- - -

Thans,

Alfredo

BenelliFurtado, 2025-03-18:
I noticed such behavior (with other ble libraries as well) that my device sometimes does no longer connect if the device is still bonded / paired with the phone and something changed on the device itself (firmware was flashed or sometimes after a DFU with another phone). Deleting the bond from the system settings helped (in my case).

riccardocescon, 2025-08-07:
This is happening to me with a Realme 14 Pro 5G - Android 15
Does this means something on the package is wrong on the native Android part?
By using other applications (such as connecting with NRF Connect) does not result in any weird connection/disconnection